robotemi/sdk #332: When I call askQuestion the first time it fails

CarlClement, 2022-09-29:
I have an application that asks a question to start a children's story. The first time I run the application the question is never asked. However if I stop the application and restart it the question is asked and ASR runs correctly. I am using version 1.129.4 of the sdk on a version 2 Temi.

Thanks

zjn0505, 2022-10-08:
I suppose it is related to a bug that the TTS engine is not properly injected in the temi launcher you used, version 16398 or 16400.

The first time of speak would fail if it is using non-system language. i.e. speak French when English is the system language.

You can try to speak() something in the target language after boot to let the engine initialized.

Please help to check if this can be used as workaround.

CarlClement, 2022-10-27:
Sorry for the delay, been heads down developing. Yes this does work as a workaround. For the record the issue I am having is not related to a non-system language, I am getting this in the system language (EN-US).

zjn0505, 2022-11-09:
I thought it would happen when you call it like 

`Robot.speak(TtsRequest.create("Hello, world!", language = TtsLanguage.EN_US))`

You mean if the system language is English, it would also happen when you call it like

`Robot.speak(TtsRequest.create("Hello, world!"))`

prashantsmp, 2022-11-09:
yes.. In my application anytime the language can be switched to other language.
So i created the TTSRequest like below,

val ttsRequest : TtsRequest = if (AppConstants.isEnglish) {
                TtsRequest.create(text, isShowOnConversationLayer = false, language = TtsRequest.Language.EN_US, showAnimationOnly = false, cached = true)
            } else {
                TtsRequest.create(text, isShowOnConversationLayer = false, language = TtsRequest.Language.AR_AE, showAnimationOnly = false, cached = true)
            }

how can we overcome this TTSEngine issue?

zjn0505, 2022-11-10:
We've fixed this on 129 version for V3 robot., also the `cached` parameter is only available from 129 version.

There may not be a hotfix for 128 version V2 robot in a short period of time.