robotemi/sdk #333: Alarm manager not working in offline mode

iamkhan001, 2022-10-05:
Hi, I need to perform certain actions at a particular time, for this, I'm using Alarm Manager to set actions of selected time.
We notice that when temi is Online it's working fine. but when we turn off wifi and test in offline mode, it's not working.
Please help or suggestions on this.

below is code for reference. 
```
alarmManager.set(
                AlarmManager.RTC_WAKEUP,
                Calendar.getInstance().apply {
                    set(Calendar.HOUR_OF_DAY, hour)
                    set(Calendar.MINUTE, minute)
                    set(Calendar.SECOND, 0)
                }.timeInMillis,
                PendingIntent.getBroadcast(
                    requireContext(),
                    scheduleId.toInt(),
                    Intent(requireContext(), AlarmBroadcastReceiver::class.java).apply {
                        putExtra("time", time)
                        putExtra("payload", payload)
                    },
                    PendingIntent.FLAG_CANCEL_CURRENT
                )
            )
```

zjn0505, 2022-10-08:
Is this on temi V2 or V3?

iamkhan001, 2022-10-12:
Temi V2

zjn0505, 2022-10-12:
Hi, I did a very simple test with an alarm set to current timestamp + 4000 and it will be triggered on time with or without internet.

I suppose even this simple test doesn't stand for a real life use case, but a fundamental Android alarm shouldn't go wrong just because of internet connection.

I wonder if robot clock may cause some problem, the tablet of temi doesn't have a battery, so its clock will be reset to an early time without a clock sync from network. Maybe this is the reason that if your robot has been offline since boot, then its clock is not synced, thus alarm is not triggered on time?