robotemi/sdk #334: Standby Screen

eSalto, 2022-10-07:
I am trying to find out when the Temi is on the standby screen. I was wondering if there are any listeners who know this. On the other hand, can I put the standby screen through the SDK?

Thanks,

zjn0505, 2022-10-08:
Standby is supported since v1.129.0 for temi V3 robot.

https://github.com/robotemi/sdk/releases/tag/v1.129.0

https://github.com/robotemi/sdk/wiki/Utils#isStandByOn
https://github.com/robotemi/sdk/wiki/Utils#startstandby
https://github.com/robotemi/sdk/wiki/Utils#stopstandby

There isn't a listener yet. Please check if this could support your needs.

Let me know if you need to have more interfaces.

zjn0505, 2025-02-11:
> > https://github.com/robotemi/sdk/wiki/Utils#isStandByOn
> > https://github.com/robotemi/sdk/wiki/Utils#startstandby
> > https://github.com/robotemi/sdk/wiki/Utils#stopstandby
> 
> Are these functions supported on Temi V2? They don't seem to be doing anything, yet they return 0.

No, standby is supported since v1.129.0 for temi V3 robot.