robotemi/sdk #335: Connect to additional web cam

MiRobotic, 2022-10-11:
Hi 

we would like to connect additional webcam with temi by using cable. However, the webcam is not able to connect to temi. Any suggestion about this?

We want to detect some "action" while robot movement. While the robot moving around, we cannot access the camera

zjn0505, 2022-10-12:
It is not easy to use external camera on temi

On temi V2, an external camera via USB can be supported only with a special Android FW, which will disable the built-in cameras.

On temi V3, an external camera would be detected by UVC protocol.

mbreslein-thd, 2023-02-22:
@zjn0505 
> On temi V2, an external camera via USB can be supported only with a special Android FW, which will disable the built-in cameras.

Could you give us more details about that? I'm also currently trying to get a external camera working.