robotemi/sdk #340: OnMovementStatusChanged

khs3994, 2022-11-05:
OnMovementStatusChanged does not work when the Temi moves or reaches the target. Is my method wrong?

```kotlin
@AndroidEntryPoint
class MovingActivity : BaseActivity<ActivityMovingBinding>(R.layout.activity_moving),
    OnMovementStatusChangedListener {

    private val mainViewModel by viewModels<MainViewModel>()

    override fun init() {
        animation()
        goTable()
        initUi()
        robotInit()
    }

    override fun onMovementStatusChanged(type: String, status: String) {
        Log.d("TemiStatus", "type: $type, status: $status")
    }

    private fun robotInit() {
        mainViewModel.robot.addOnMovementStatusChangedListener(this)
        mainViewModel.robot.hideTopBar()
    }

    private fun initUi() {
        binding.tableNum.text = intent.getStringExtra("seatNum")
    }

    private fun animation() {
        val anim = AnimationUtils.loadAnimation(this, R.anim.alpha)
        val anim2 = AnimationUtils.loadAnimation(this, R.anim.alpha2)
        val anim3 = AnimationUtils.loadAnimation(this, R.anim.alpha3)

        val goFront = ObjectAnimator.ofFloat(binding.temi, "translationX", 480f).apply {
            duration = 3000
            repeatCount = ValueAnimator.INFINITE
        }

        with(binding) {
            exDot1.animation = anim
            exDot2.animation = anim2
            exDot3.animation = anim3
        }
        goFront.start()
    }

    private fun goTable() {
        mainViewModel.robot.goTo(intent.getStringExtra("seatNum")!!)
    }
}
```

zjn0505, 2022-11-05:
For goto, this listener shall be used. https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener.

onMovementStatusChangedListener will provide status changes of manual control movement, including [skidjoy](https://github.com/robotemi/sdk/wiki/Movement#skidJoy) control and [turnBy]https://github.com/robotemi/sdk/wiki/Movement#turnBy()

khs3994, 2022-11-06:
> For goto, this listener shall be used. https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener.
>
> onMovementStatusChangedListener will provide status changes of manual [...]

Thank's 😀

closed 2022-11-09 by zjn0505