robotemi/sdk #346: Confirming if Multi-Floor support is available for Temi V1

lohwenxin, 2022-11-23:
Hello, 

I am raising this issue as I am trying to setMultiFloorEnabled for Temi V1.

When I call the [setMultiFloorEnabled](https://github.com/robotemi/sdk/wiki/Locations#setMultiFloorEnabled) method and print isMultiFloorEnabled, I get a null result.
[image]
[image]

As per the requirements of the setMultiFloorEnabled method, I have MAP and SETTINGS permissions and my temi version is greater than 1.129.0. 
[image]
[image]

The Multi-Floor support was included in the 1.129.0 release, but I noted that the release says that this version is only applicable for Temi V3. **Could I re-confirm that there is no Multi-Floor support for Temi V1?** Subsequently, may I check if support for Multi-Floor will be available for Temi V1 and V2 in further releases?
[image]

Thank you!
Wen Xin

zjn0505, 2022-11-23:
Major version updates won't be available for temi V1 and V2, only hotfixes upon 128 version for potential critical bugs and broken features would be provided.

Most of the changes in temi SDK will require a comptible version of temi launcher, multifloor is for V3 robot on 129 version.

lohwenxin, 2022-11-24:
Alright, thank you for the clarification :+1:

closed 2022-11-25 by zjn0505