robotemi/sdk #348: Temi Answering Questions outside application

prashantsmp, 2022-11-25:
I’m using`robot.askQuestion(“which floor you want to go?” ` user replied as 3. onAsrResult() method I can able to get the answer as 3 but after few seconds temi fetches answer from outside the application and answering for 3 and changed volume to high. During this there’s one icon appears on top bar like skill set. How can allow my application to answer users questions inside my application alone?

zjn0505, 2022-11-25:
temi will invoke an NLP query when user answers the question.

If you want to cut the flow, you will have to override default NLP.

prashantsmp, 2022-11-25:
> temi will invoke an NLP query when user answers the question.
> 
> If you want to cut the flow, you will have to override default NLP.

How to enable the override default NLP?

zjn0505, 2022-11-25:
https://github.com/robotemi/sdk/wiki/Speech#override-the-nlp-

This is the way to override NLP. but it may not suit your needs.

prashantsmp, 2022-11-25:
> https://github.com/robotemi/sdk/wiki/Speech#override-the-nlp-
> 
> This is the way to override NLP. but it may not suit your needs.

What will be the correct approach?