robotemi/sdk #363: Temi screent tilt before move

Magiccofee, 2023-01-25:
I was trying to have Temi navigate along a series of waypoints using goToPosition(). During the testing, I noticed that Temi will always spend some time to tilt its screen before moving. Sometimes the robot would even move a bit before jerking to a stop to tilt the screen before continuing. 

Is there a way to disable this behavior in order to speed up the move process?

Also is offline map loading supported by Temi V2?

Tested on a Temi V2, using SDK Version 0.10.80

zjn0505, 2023-01-30:
temi is using the depth camera on the tablet to avoid obstacles duing go to location. So it will tilt and look to the front before moving.

This tilt behavior cannot be diabled right now. and cannot be speed up.

Also when go to a location, it will tilt to angles when the location was saved, goToPosition doesn't have this extra movement.

If your target robots will include temi V3, you can also check this api https://github.com/robotemi/sdk/wiki/Movement#patrol, it can go between multiple locations with a few parameters to fine control it.

V2 also have offline map cache, it would cache up to 32 maps (or 400 MB of total map size) when they are exported to and imported from the cloud.

But before 129 version, this cache is self-managed black box and cannot be controled from SDK, it was mean to help to shorten the time of map loading. If you load map offline, it would work when the target map was cached. There will be a little time to wait for a network request timeout when it try to check latest map data of target map. If it timeouts, then it would fallback to local cache.

Only in 129 version it provides methods to prefetch maps to cache, and force offline map loading.

Magiccofee, 2023-01-30:
Hi, thank you for the response. Regarding the depth camera, is there anyway to receive the feed and use it for other tasks. Also, I understand that in 129 version there is the method to force offline map loading. Can I check if that function is available to Temi with Launcher OS 16400, Robox OS 128.12 and firmware 20201216.154352. If not, is there a way for Temi to maintain network connectivity but

Magiccofee, 2023-01-30:
Sorry about that, I accidentally clicked on the close button.

Magiccofee, 2023-01-30:
Hi, thank you for the response. Regarding the depth camera, is there anyway to receive the feed and use it for other tasks?

Also, I understand that in 129 version there is the method to force offline map loading. Can I check if that function is available to Temi with Launcher OS 16400, Robox OS 128.12 and firmware 20201216.154352?

If not, is there a way for Temi to maintain network connectivity but have the network check to the cloud service be a timeout? I feel like an offline map load would be faster but the Temi has to stay connected to receive new move commands.

Thank you.

zjn0505, 2023-02-01:
We don't expose the raw data of depth camera to SDK and even to Android. Its data is processed exclusively in the Robox, temi's algorithm core.

On the Android we only utilize Robox's transformed data for user detection, for example in greet mode.

https://github.com/robotemi/sdk/wiki/Detection-&-Interaction#setdetectionmodeon

You can set distance and listen to the detection results. It is powered by depth camera behind the scenes.

Force offline map loading is only provided from 129. Launcher OS is 16892 and above. The cloud saved maps can be big and small, we haven't fine tuned the timeout when fetching cloud saved map. Even we do it, it would only be available in newer releases.

Magiccofee, 2023-02-02:
Alright. Thank you for the clarifications.

closed 2023-02-02 by Magiccofee