robotemi/sdk #369: Replacing TEMI's NLP model with our own custom-built model

aiyunguo, 2023-02-15:
Hi, noticed that there is an option to override the NLP in the documentations. Would like to find out more on how to replace the default NLP model in TEMI to our own custom defined model. For instance, could we fit in our own pre-trained NLP (e.g. GPT2, ChatGPT) model to process the text input instead of using TEMI's default one?

Thank you!

prashantsmp, 2023-09-12:
Did you find any solutions?

zjn0505, 2023-09-15:
Yes, a few steps to take.

1. Declare NLP overriden in XML
2. Declare and request to be the selected kiosk app.

Given these steps, after temi wake up, it will still transcript voice into text. But it don't proceed with temi NLP.

Then kiosk app can take the ASR result from onAsrResult() callback, and forward it to your own NLP.

If your app cannot take the intent, like request to save locations, then your can still call startDefaultNlu(text = asrResults), meaning give the asr back to temi NLP to process.

We have some clients and developers already did the override in their app, and conver it ChatGPT or other NLP services.

https://market.robotemi.com/details/robocoregpt

prashantsmp, 2023-10-05:
Can we get some details about? It will be very helpful for the other developers to break a leg?