robotemi/sdk #374: Wakeup not working

moritzsu, 2023-03-08:
Sometimes when I wakeup temi programmaticaly via robot.wakeup() this does not work correctly.
I get two onConversationStatusChanged events and after that nothing anymore, temi also does not listen for speech.
A short sound is also played when that happens and it hapens often when wifi signal is lost for a short amount of time.

Any ideas/suggestions on how I can fix that?

moritzsu, 2023-03-09:
After further investigation the time apart between the two OnConversationStatusChanged Events also differ (IDLE and LISTENING)
Sometimes they are 5 seconds apart (case where everything is good) and sometimes only a few  milliseconds (case where it does not work), is this the cause for the problem?

zjn0505, 2023-03-13:
The short sound is an indication of network connection.

The speech to text engine is backed by Google ASR engine, and it requires sending streaming requests to their clould.

moritzsu, 2023-03-13:
Is there a way that I can check if the wakeup() command was successful then?
I am calling wakeup() in a loop so that temi is always listening and this is disrupting the loop rather often.