robotemi/sdk #381: Enable Temi Listening Mode

prashantsmp, 2023-03-22:
I am creating a chatbot application, where i need to detect the user and ask his input query to fetch from the database.

In order to get input from the user through ASR, we need to enable kiosk mode. during this kiosk mode temi robot's wakeup function not working. i couldn't able to trigger wakeup method for the user input.

Kindly let me know, the temi robot always enabled for the user's input.

monchisss, 2023-03-24:
can you share how yo do the chatbot app please ? we are trying to do the same but we dont have a clue how to do it

onebiwebdev, 2023-03-26:
> can you share how yo do the chatbot app please ? we are trying to do the same but we dont have a clue how to do it

You can use our app:
https://market.robotemi.com/details/one-chatbot

Based on GPT3

github-ask, 2023-05-01:
> > can you share how yo do the chatbot app please ? we are trying to do the same but we dont have a clue how to do it
> 
> You can use our app: https://market.robotemi.com/details/one-chatbot
> 
> Based on GPT3

How can I show the face (animation) by a code while it's not speaking similar to how it's in your application?

prashantsmp, 2023-08-04:
> How can I show the face (animation) by a code while it's not speaking similar to how it's in your application?

I do have this query for long time. 
They alway marketing in media like temi chatting with face animation with the user, but in SDK there are no way. Sad Reality.