robotemi/sdk #382: Add Contacts by SDK

eSalto, 2023-03-31:
Hi, that contacts can be added by SDK? If not, what do you plan to add in future versions?

thanks

zjn0505, 2023-04-03:
There is no plan to add it yet.

Can you describe your use case? Maybe we can help in another way.

eSalto, 2023-04-05:
In our use case we want to be able to add contacts through our application. Right now we show the list of contacts that Temi can call, but if the user wants to add new contacts, he has to let us know before so we can add them from the temi center.

zjn0505, 2023-04-18:
We don't have this in the design yet, and probably won't allow apps to directly modify temi or user resource in the future.

There is a definition of role in temi's organization, you may add the temi users as Collaborator in the organization, so they will only have access to robots they were delegated to with appropriate permissions.

eSalto, 2023-04-24:
All right, we'll close the issue

closed 2023-04-24 by eSalto