robotemi/sdk #387: Access the onboard camera

mblees, 2023-05-05:
Hey there,
I want to develop a feature which allows me to use the camera to scan QR-Codes. It will be used to bring visitors to their desired location for a meeting.
Is there a functionality for accessing the feed from the camera or is there any other way to implement my problem?
I would be thankfull for any help.
Poirot

zjn0505, 2023-05-07:
It is Android cameras, be accessed from Android native camera API.

jay-m-dev, 2023-10-05:
@mblees were you able to build an app with this functionality?

mblees, 2023-10-05:
> @mblees were you able to build an app with this functionality?

Yes i was able to use CameraX to access the Camera after a lot of issues while implementing

vincenzomarotta, 2024-09-10:
> > @mblees were you able to build an app with this functionality?
> 
> Yes i was able to use CameraX to access the Camera after a lot of issues while implementing

Hi, can I contact you? I have to access temi's camera but I'm having some problems

vincenzomarotta, 2024-09-10:
> @vincenzomarotta Feel free to add me on discord maybe i can help you out: rgfpoirot

Thanks 🙏🏻 I've sent you a request as lookatviki