robotemi/sdk #394: map present in themes in the app

fabiola29, 2023-06-14:
Hi I would like to port the app present in themes into my app I am trying the sample code of getDataMap but I have this error:
E/Robot: No files supported by provider at content://com.robotemi.sdk.provider/map_data_image?mediaKey=
how can i fix it?

zjn0505, 2023-06-16:
Hi, did you run the sample app and checked the map from RESOURCES -> Get Map Data?

I also tested getMapData API and there is not problem to get the map.
[image]
[image]

fabiola29, 2023-06-16:
maybe I have to accept some permission?