robotemi/sdk #404: Talk about topics

fabiola29, 2023-07-21:
Hi, I'm making an app where when opening Temi something must fire and when it switches to the other screen or when it returns to the starting screen it doesn't say anything I tried to insert this                                                                                                                 

```
robot.speak(TtsRequest.create("Hello, my name's is Themes", false)); 
Robot.getInstance().cancelAllTtsRequests();    
```
 in onRobotReady but I don't have what I want, i.e. when I switch to the screen and return to the starting one, the theme speaks again, how can I solve it?

prashantsmp, 2023-08-01:
Can you please elaborate your question?

fabiola29, 2023-08-04:
> Can you please elaborate your question?

the app opens and themes says something, then it goes to another screen and it doesn't say anything when I then go back to the screen where themes says something the message from themes is repeated I want I'm on a screen and themes says something when I then return at that screen themes says nothing, I hope I have clarified :)

prashantsmp, 2023-08-04:
> Hi, I'm making an app where when opening Temi something must fire and when it switches to the other screen or when it returns to the starting screen it doesn't say anything I tried to insert this [...]

You just need to change your code like below,
Robot.getInstance().cancelAllTtsRequests();  // It will clear previous speech content added in the queue.
robot.speak(TtsRequest.create("Hello, my name's is Themes", false));  It will speak the content which you provided.

If its not helps you, then please share your activity class.

fabiola29, 2023-08-04:
> You just need to change your code like below, Robot.getInstance().cancelAllTtsRequests(); // It will clear previous speech content added in the queue. robot.speak(TtsRequest.create("Hello, my name's [...]

but maybe I'm wrong in the position where they should be inserted, where should I insert them? because I tried in the onCreate and nothing happens.

zjn0505, 2023-08-07:
Maybe you have added the onRobotReady listener inside onStart or onResume, so it is registered every time when activity is resumed.

As I remember when you register the listener, onRobotReady will be called immediately if it ready, so you may get this callback too often.

fabiola29, 2023-08-07:
> Maybe you have added the onRobotReady listener inside onStart or onResume, so it is registered every time when activity is resumed.
> 
> As I remember when you register the listener, onRobotReady will be called immediately if it ready, so you may get this callback too often.

I inserted here: 

```
` @Override
    public void onRobotReady(boolean isReady) {
        if (isReady) {
            try {

                Robot.getInstance().cancelAllTtsRequests();
                robot.speak(TtsRequest.create("Hello, my name's is Themes", false));
                final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);
                robot.onStart(activityInfo);

            } catch (PackageManager.NameNotFoundException e) {
                throw new RuntimeException(e);
            }
        }
    }`
```

and as onStart I have ho:

```
 /**
     * Setting up all the event listeners
     */
    @Override
    public void onStart() {
        super.onStart();

        // Add robot event listeners
        robot.addOnRobotReadyListener(this);
    }

    /**
     * Removing the event listeners upon leaving the app.
     */
    @Override
    public void onStop() {
        super.onStop();

        // Remove robot event listeners
        robot.removeOnRobotReadyListener(this);
    }
```
the onResume it's not on the screen

prashantsmp, 2023-08-07:
> I inserted here:
>
> ```
> ` @Override
>     public void onRobotReady(boolean isReady) {
>         if (isReady) {
>             try { [...]

I would like you to read the documentation first, then read the sample code’s  main activity. 

Add your code below,

```

 final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);                 robot.onStart(activityInfo);

robot.speak(TtsRequest.create("Hello, my name's is Themes", false));

```

fabiola29, 2023-08-07:
> I would like you to read the documentation first, then read the sample code’s main activity.
>
> Add your code below,
>
> ```
>
>  final ActivityInfo activityInfo = [...]

so just do this
```
` @Override
    public void onRobotReady(boolean isReady) {
        if (isReady) {
            try {

          
                final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);
                robot.onStart(activityInfo);
             
                Robot.getInstance().cancelAllTtsRequests();  // should i put it?
                robot.speak(TtsRequest.create("Hello, my name's is Themes", false));

            } catch (PackageManager.NameNotFoundException e) {
                throw new RuntimeException(e);
            }
        }
    }`

```

zjn0505, 2023-08-10:
Here I will rephrase my last response to make it clearer.

Everytime your app goes back to the starting screen, `onStart()` block of your fragment will be triggered.

Your app is adding this listener  `robot.addOnRobotReadyListener(this);` in `onStart()` callback.

You can check this in SDK source code. When you add listener, you will be notified immediately about robot readiness.

https://github.com/robotemi/sdk/blob/master/sdk/src/main/java/com/robotemi/sdk/Robot.kt#L792

So that your page's `onRobotReady` callback will be triggered.

In `onRobotReady` callback, you app will call `robot.speak(TtsRequest.create("Hello, my name's is Themes", false));`

So it speaks every time when your app goes back to the starting screen.

----

I think the quick solution will be adding a boolean to only speak it once.

```java

    @Override
    public void onRobotReady(boolean isReady) {
        if (isReady) {
            try {

          
                final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);
                robot.onStart(activityInfo);
                if (!((MainAcitivity) getActivity()).ttsSpoken) {
                    Robot.getInstance().cancelAllTtsRequests();  // should i put it?
                    robot.speak(TtsRequest.create("Hello, my name's is Themes", false));
                    ((MainAcitivity) getActivity()).ttsSpoken = true;
                }

            } catch (PackageManager.NameNotFoundException e) {
                throw new RuntimeException(e);
            }
        }
    }`
    ```

fabiola29, 2023-08-10:
> Here I will rephrase my last response to make it clearer.
> 
> Everytime your app goes back to the starting screen, `onStart()` block of your fragment will be triggered.
> 
> Your app is adding this listener `robot.addOnRobotReadyListener(this);` in `onStart()` callback.
> 
> You can check this in SDK source code. When you add listener, you will be notified immediately about robot readiness.
> 
> https://github.com/robotemi/sdk/blob/master/sdk/src/main/java/com/robotemi/sdk/Robot.kt#L792
> 
> So that your page's `onRobotReady` callback will be triggered.
> 
> In `onRobotReady` callback, you app will call `robot.speak(TtsRequest.create("Hello, my name's is Themes", false));`
> 
> So it speaks every time when your app goes back to the starting screen.
> 
> I think the quick solution will be adding a boolean to only speak it once.
> 
> ```java
>     @Override
>     public void onRobotReady(boolean isReady) {
>         if (isReady) {
>             try {
> 
>           
>                 final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);
>                 robot.onStart(activityInfo);
>                 if (!((MainAcitivity) getActivity()).ttsSpoken) {
>                     Robot.getInstance().cancelAllTtsRequests();  // should i put it?
>                     robot.speak(TtsRequest.create("Hello, my name's is Themes", false));
>                     ((MainAcitivity) getActivity()).ttsSpoken = true;
>                 }
> 
>             } catch (PackageManager.NameNotFoundException e) {
>                 throw new RuntimeException(e);
>             }
>         }
>     }
> ```
> 
> If it fits your app, you can also register onRobotReady listener in the activity, and let it coordinate the events for fragments

ok but this variable `public boolean ttsSpoken = false; `I have to put it in the class where I have this :  
```
@Override
    public void onRobotReady(boolean isReady) {
        if (isReady) {
            try {

          
                final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);
                robot.onStart(activityInfo);
                if (!((MainAcitivity) getActivity()).ttsSpoken) {
                    Robot.getInstance().cancelAllTtsRequests();  // should i put it?
                    robot.speak(TtsRequest.create("Hello, my name's is Themes", false));
                    ((MainAcitivity) getActivity()).ttsSpoken = true;
                }

            } catch (PackageManager.NameNotFoundException e) {
                throw new RuntimeException(e);
            }
        }
    }
``` 
because I have an activityMain that I use for permissions and then I have the HomeFragmet where I have to do the above.

fabiola29, 2023-08-10:
Specifically I have this:
```
public class MainActivity extends AppCompatActivity {

    private static final int REQUEST_EXTERNAL_STORAGE = 1;
    private final static String[] PERMISSIONS_STORAGE = {Manifest.permission.READ_EXTERNAL_STORAGE, Manifest.permission.WRITE_EXTERNAL_STORAGE};
    public boolean ttsSpoken = false;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        GlobalViewModel viewModel = new ViewModelProvider(this).get(GlobalViewModel.class);

        viewModel.initialize();

        setContentView(R.layout.activity_main);
        verifyStoragePermissions(this);
    }

    /**
     * Checks if the app has permission to write to device storage
     * If the app does not has permission then the user will be prompted to grant permissions
     */
    public static void verifyStoragePermissions(Activity activity) {
        // Check if we have write permission
        int permission = ActivityCompat.checkSelfPermission(activity, Manifest.permission.WRITE_EXTERNAL_STORAGE);
        if (permission != PackageManager.PERMISSION_GRANTED) {
            // We don't have permission so prompt the user
            ActivityCompat.requestPermissions(activity, PERMISSIONS_STORAGE, REQUEST_EXTERNAL_STORAGE);
        }
    }
}
```
and : 
```
public class HomeFragment extends Fragment implements OnRobotReadyListener{

    private Robot robot;
    private CardView bt_create_route,bt_start_patrol,bt_option_location;
    private ImageButton bt_back;
    private NavController navController;
    private GlobalViewModel viewModel;

    /**
     * Setting up all the event listeners
     */
    @Override
    public void onStart() {
        super.onStart();

        // Add robot event listeners
        robot.addOnRobotReadyListener(this);
    }

    /**
     * Removing the event listeners upon leaving the app.
     */
    @Override
    public void onStop() {
        super.onStop();

        // Remove robot event listeners
        robot.removeOnRobotReadyListener(this);
    }

    @Override
    public View onCreateView(@NotNull LayoutInflater inflater,
                             ViewGroup container,
                             Bundle savedInstanceState) {
        viewModel = new ViewModelProvider(getActivity()).get(GlobalViewModel.class);
[... 69 lines cut ...]
            }
        });
    }

    @Override
    public void onRobotReady(boolean isReady) {
        if (isReady) {
            try {

                final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);
                robot.onStart(activityInfo);
                if (!((MainActivity) getActivity()).ttsSpoken) {
                    Robot.getInstance().cancelAllTtsRequests();
                    robot.speak(TtsRequest.create("Hello, my name's is Themes", false));
                    ((MainActivity) getActivity()).ttsSpoken = true;
                }
            } catch (PackageManager.NameNotFoundException e) {
                throw new RuntimeException(e);
            }
        }
    }

}
```
Is correct?

zjn0505, 2023-08-11:
Yes. Something like that. 

Or I suppose you can use the GlobalViewModel to keep it.

prashantsmp, 2023-08-11:
I would suggest you to implement OnRobotReadyListener in MainActivity, then you can access robot instance all over the fragment. 

If you're using multiple fragments in future, then you need to initialize for each & every fragments. It's not a ideal solution to implement on Fragment.  

You can use single activity and multiple fragments. 

In Main Activity, you can implement this, then your problem reinitiating everytime will be solved.

    @Override
    public void onCreate() {
        super.onCreate;
        // Add robot event listeners
        robot.addOnRobotReadyListener(this);
    }

    
    @Override
    public void onDestroy() {
        super.onDestroy();
        // Remove robot event listeners
        robot.removeOnRobotReadyListener(this);
    }
```

fabiola29, 2023-08-11:
> I would suggest you to implement OnRobotReadyListener in MainActivity, then you can access robot instance all over the fragment.
>
> If you're using multiple fragments in future, then you need to [...]

ok so i insert this in the activity main and in the fragmentHome i leave it like this```
 ```
public class HomeFragment extends Fragment implements OnRobotReadyListener{

    private Robot robot;
    private CardView bt_create_route,bt_start_patrol,bt_option_location;
    private ImageButton bt_back;
    private NavController navController;
    private GlobalViewModel viewModel;

    /**
     * Setting up all the event listeners
     */
    @Override
    public void onStart() {
        super.onStart();

        // Add robot event listeners
        robot.addOnRobotReadyListener(this);
    }

    /**
     * Removing the event listeners upon leaving the app.
     */
    @Override
    public void onStop() {
        super.onStop();

        // Remove robot event listeners
        robot.removeOnRobotReadyListener(this);
    }

    @Override
    public View onCreateView(@NotNull LayoutInflater inflater,
                             ViewGroup container,
                             Bundle savedInstanceState) {
        viewModel = new ViewModelProvider(getActivity()).get(GlobalViewModel.class);

        return inflater.inflate(R.layout.fragment_home, container, false);
    }

    @Override
    public void onViewCreated(@NonNull final View view, Bundle savedInstanceState) {
        super.onViewCreated(view, savedInstanceState);
        navController = Navigation.findNavController(view);

        ((AppCompatActivity) getActivity()).getSupportActionBar().hide();

        getActivity().getWindow().setSoftInputMode(WindowManager.LayoutParams.SOFT_INPUT_STATE_HIDDEN);

        bt_create_route = view.findViewById(R.id.cardView_create_route);
        bt_create_route.setOnClickListener(v -> button_create_route());

        bt_start_patrol = view.findViewById(R.id.cardView_start_patrol);
        bt_start_patrol.setOnClickListener(v -> button_start_patrol());

        bt_option_location = view.findViewById(R.id.cardView_option_location);
        bt_option_location.setOnClickListener(v -> button_option_location());

        bt_back =  view.findViewById(R.id.btn_back);
        bt_back.setOnClickListener(v -> button_back());

        robot = Robot.getInstance();  // get an instance of the robot in order to begin using its features.

        // Robot.getInstance().cancelAllTtsRequests();
        // robot.speak(TtsRequest.create("Hello, my name's is Temi", false));
    }

    private void button_back(){
        bt_back.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                getActivity().finish();
            }
        });
    }

        private void button_create_route(){
        bt_create_route.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                GlobalViewModel viewModel = new ViewModelProvider(getActivity()).get(GlobalViewModel.class);
                viewModel.initializeForRouteCreation();
                navController.navigate(R.id.action_homeFragment_to_createRouteFragment);
            }
        });
    }

    private void button_start_patrol(){
        bt_start_patrol.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                GlobalViewModel viewModel = new ViewModelProvider(getActivity()).get(GlobalViewModel.class);
                viewModel.initializeForRouteCreation();
                navController.navigate(R.id.action_homeFragment_to_routeExecutionFragment);
            }
        });
    }

    private void button_option_location(){
        bt_option_location.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                GlobalViewModel viewModel = new ViewModelProvider(getActivity()).get(GlobalViewModel.class);
                viewModel.initializeForRouteCreation();
                navController.navigate(R.id.action_HomeFragment_to_optionLocationFragment);
            }
        });
    }

    @Override
    public void onRobotReady(boolean isReady) {
        if (isReady) {
            try {

                final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);
                robot.onStart(activityInfo);
                if (!((MainActivity) getActivity()).ttsSpoken) {
                    Robot.getInstance().cancelAllTtsRequests();
                    robot.speak(TtsRequest.create("Hello, my name's is Themes", false));
                    ((MainActivity) getActivity()).ttsSpoken = true;
                }
            } catch (PackageManager.NameNotFoundException e) {
                throw new RuntimeException(e);
            }
        }
    }
}
```
?

prashantsmp, 2023-08-14:
Implements OnRobotReadyListener in MainActivity and remove it from HomeFragment.

So in MainActivity,  you will have this override method,
```
//Global Variable
private Robot robot; 

@Override
   public void onRobotReady(boolean isReady) {
       if (isReady) {
           try {
               final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);
               robot.onStart(activityInfo);
               robot = this
           } catch (PackageManager.NameNotFoundException e) {
               throw new RuntimeException(e);
           }
       }
    }
```
  

Create one function in MainActivity, 

```
public void robotToSpeak(String speechText) {
  if (robot != null) {
     robot.cancelAllTtsRequests();
     robot.speak(TtsRequest.create(speechText, false));
 }
}
```

If you want access this method from any Fragments then you need to call like below,

```
if ( getActivity() != null) {
   ((MainActivity) getActivity()).robotToSpeak("Hello, my name's is Temi")
}
```

I hope you can follow this & solve your problems.

fabiola29, 2023-08-14:
> Implements OnRobotReadyListener in MainActivity and remove it from HomeFragment or any other fragments you already have implemented.
> 
> So in MainActivity, you will have this override method,
> 
> ```
> //Global Variable
> private Robot robot; 
> 
> 
> @Override
>    public void onRobotReady(boolean isReady) {
>        if (isReady) {
>            try {
>                final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);
>                robot.onStart(activityInfo);
>                robot = this
>            } catch (PackageManager.NameNotFoundException e) {
>                throw new RuntimeException(e);
>            }
>        }
>     }
> ```
> 
> Create one function in MainActivity,
> 
> ```
> public void robotToSpeak(String speechText) {
>   if (robot != null) {
>      robot.cancelAllTtsRequests();
>      robot.speak(TtsRequest.create(speechText, false));
>  }
> }
> ```
> 
> If you want access this method from any Fragments then you need to call like below,
> 
> ```
> if ( getActivity() != null) {
>    ((MainActivity) getActivity()).robotToSpeak("Hello, my name's is Temi")
> }
> ```
> 
> I hope you can follow this & solve your problems.
> 
> After doing the above changes, your home fragment will be look like below,
> 
> ```
> public class HomeFragment extends Fragment {
> 
>     private CardView bt_create_route,bt_start_patrol,bt_option_location;
>     private ImageButton bt_back;
>     private NavController navController;
>     private GlobalViewModel viewModel;
> 
>     @Override
>     public View onCreateView(@NotNull LayoutInflater inflater,
>                              ViewGroup container,
>                              Bundle savedInstanceState) {
>         viewModel = new ViewModelProvider(getActivity()).get(GlobalViewModel.class);
> 
>         return inflater.inflate(R.layout.fragment_home, container, false);
>     }
> 
> 
>     @Override
>     public void onViewCreated(@NonNull final View view, Bundle savedInstanceState) {
>         super.onViewCreated(view, savedInstanceState);
>         navController = Navigation.findNavController(view);
> 
>         ((AppCompatActivity) getActivity()).getSupportActionBar().hide();
> 
>         getActivity().getWindow().setSoftInputMode(WindowManager.LayoutParams.SOFT_INPUT_STATE_HIDDEN);
> 
>         bt_create_route = view.findViewById(R.id.cardView_create_route);
>         bt_create_route.setOnClickListener(v -> button_create_route());
> 
>         bt_start_patrol = view.findViewById(R.id.cardView_start_patrol);
>         bt_start_patrol.setOnClickListener(v -> button_start_patrol());
> 
>         bt_option_location = view.findViewById(R.id.cardView_option_location);
>         bt_option_location.setOnClickListener(v -> button_option_location());
> 
>         bt_back =  view.findViewById(R.id.btn_back);
>         bt_back.setOnClickListener(v -> button_back());
> 
>         if ( getActivity() != null) {
>            ((MainActivity) getActivity()).robotToSpeak("Hello, my name's is Temi")
>         }
>         
>     }
> 
> 
>     private void button_back(){
>         bt_back.setOnClickListener(new View.OnClickListener() {
>             @Override
>             public void onClick(View v) {
>                 getActivity().finish();
>             }
>         });
>     }
> 
>         private void button_create_route(){
>         bt_create_route.setOnClickListener(new View.OnClickListener() {
>             @Override
>             public void onClick(View v) {
>                 GlobalViewModel viewModel = new ViewModelProvider(getActivity()).get(GlobalViewModel.class);
>                 viewModel.initializeForRouteCreation();
>                 navController.navigate(R.id.action_homeFragment_to_createRouteFragment);
>             }
>         });
>     }
> 
> 
>     private void button_start_patrol(){
>         bt_start_patrol.setOnClickListener(new View.OnClickListener() {
>             @Override
>             public void onClick(View v) {
>                 GlobalViewModel viewModel = new ViewModelProvider(getActivity()).get(GlobalViewModel.class);
>                 viewModel.initializeForRouteCreation();
>                 navController.navigate(R.id.action_homeFragment_to_routeExecutionFragment);
>             }
>         });
>     }
> 
> 
>     private void button_option_location(){
>         bt_option_location.setOnClickListener(new View.OnClickListener() {
>             @Override
>             public void onClick(View v) {
>                 GlobalViewModel viewModel = new ViewModelProvider(getActivity()).get(GlobalViewModel.class);
>                 viewModel.initializeForRouteCreation();
>                 navController.navigate(R.id.action_HomeFragment_to_optionLocationFragment);
>             }
>         });
>     }
> 
> }
> ```

ok perfect, but in the mainActivity I have to put this:
```

  /**
     * Setting up all the event listeners
     */
    @Override
    public void onStart() {
        super.onStart();

        // Add robot event listeners
        robot.addOnRobotReadyListener(this);
    }

    /**
     * Removing the event listeners upon leaving the app.
     */
    @Override
    public void onStop() {
        super.onStop();

        // Remove robot event listeners
        robot.removeOnRobotReadyListener(this);
    }
```
e nell'onCreate this :    `   robot = Robot.getInstance(); `

prashantsmp, 2023-08-14:
Yes you can, but onRobotReadyListener i have already assigned  **robot = this**. it is equivalent to this **robot = Robot.getInstance();**

```
@Override
   public void onRobotReady(boolean isReady) {
       if (isReady) {
           try {
               final ActivityInfo activityInfo = getActivity().getPackageManager().getActivityInfo(getActivity().getComponentName(), PackageManager.GET_META_DATA);
               robot.onStart(activityInfo);
               robot = this // Robot Instance
           } catch (PackageManager.NameNotFoundException e) {
               throw new RuntimeException(e);
           }
       }
    }
```

fabiola29, 2023-08-14:
> Yes you can, but onRobotReadyListener i have already assigned **robot = this**. it is equivalent to this **robot = Robot.getInstance();**
>
> ```
> @Override
>    public void onRobotReady(boolean isReady) [...]

okay i'm using java it's code **robot = this** gives me error so i removed it and added **robot = Robot.getInstance();**

prashantsmp, 2023-08-14:
Good. So your code is working fine as you expected?

fabiola29, 2023-08-14:
> Good. So your code is working fine as you expected?

I can't know because I don't have the robot physically at home but in the laboratory :(