robotemi/sdk #405: Serial Communication on Temi Robot

prashantsmp, 2023-08-01:
I have installed temi latest sample project on temi v3 and connected Arduino board to temi through USB port. I want to read the data from the Arduino using serial communication. SerialActivity's `override fun onSerialRawData(data: ByteArray) {}` method triggered while Arduino transmitting data via serial communication. 

My ultimate aim is to read any external serial communication data via SerialActivity's `override fun onSerialRawData(data: ByteArray) {}` method.

Is it possible?

zjn0505, 2023-08-01:
SerialActivity is created as the example for developers to read and control sensors and hardwares on temi Go robots.

https://github.com/robotemi/sdk/wiki/Serial-Communication

Developers don't need to require USB permission, integrate USB-Serial library, know the I/O protocols, they just control with one method and listen to one callback, to talk to everything newly added in temi Go.

So that SerialActivity is not designed to take other external USB-serial inputs, and it is sololy aimed for temi Go serial inputs.

----

For you project, you can take a look on this lib https://github.com/mik3y/usb-serial-for-android. This lib supported all the heavy lifting works we did for Usb-serial communication in temi Go. I suppose you can use it to control Arduino when it is connected to temi USB.

prashantsmp, 2023-08-04:
Thanks for your reply. Appreciated it. 
I have implemented this [library](https://github.com/felHR85/UsbSerial) and it fulfill my requirements.

closed 2023-08-10 by zjn0505