robotemi/sdk #411: Exception while creating a local map backup

chunpantai, 2023-09-14:
Hi, I am trying to create a local map backup using `getCurrentMapBackupFile(withoutUI: Boolean): ParcelFileDescriptor?` following the sample code.

`
ParcelFileDescriptor parcelFileDescriptor = Robot.getCurrentMapBackupFile(withoutUI = true);
`

Then I got the following exception using try catch:

`java.io.FileNotFoundException: No files supported by provider at content://com.robotemi.sdk.provider/map_backup_file?mediaKey=true`

Appreciate if anyone knows the cause and how to fix it. Thank you.

zjn0505, 2023-09-15:
Please check the following things.

1. Your temi launcher is running at version 17579 or higher.
2. Your robot has a valid PRO subscription 
3. Your app has delared and requested for MAP permission.

It is better to catch the error. Thank you for the report.

chunpantai, 2023-09-18:
Thanks for the quick response.  Currently our temi launcher is running at version 17559, could you please tell me how to update the launcher version as there is no response when I clicked the "Launcher OS" bar and it said there is no pending updates when I clicked "check for update". Very appreciate your help.

zjn0505, 2023-09-19:
Please share me your serial number. I will push and update to your robot.

chunpantai, 2023-09-19:
Due to our company policy, sorry that we cannot disclose the serial number in public. Could you please share your email contact for further correspondence? Thank you.

zjn0505, 2023-09-19:
https://robotemi.zendesk.com/

Please submit a ticket at zendesk with your serial number.