robotemi/sdk #412: Improving speech recognition especially in noisier environments

evilmtv, 2023-09-17:
Hi, speech recognition works well in ideal environments but usually have trouble in noisier environments.

Thinking of these potential solutions and would like to check if its feasible:

- Possible to install Google Services to use Google Assistant's ASR?
- Possible to use the USB C port or bluetooth to connect to an external microphone?
- Any recommend way to stream the microphone input from Temi to an external server to process it?

prashantsmp, 2023-10-25:
@evilmtv FYI, Google services not supported by Temi robot.

evilmtv, 2024-01-14:
> @evilmtv FYI, Google services not supported by Temi robot.

I'm able to get google services on Temi and use the Android/Google's built-in ASR

evilmtv, 2024-01-14:
Updates:

- Possible to install Google Services to use Google Assistant's ASR?
Yes, works well
- Possible to use the USB C port or bluetooth to connect to an external microphone?
Yes, both USB C and A ports on Temi v3 can connect to external speakers/microphones. However, both input & output will be switched to the external audio device and there is no way to programmatically switch between built-in speakers/microphone or the external one