robotemi/sdk #418: ADB connection - Unable to connect

acivit, 2023-10-23:
Hello!

I was using the ADB connection yesterday, using both terminal command (curl -k ...) and the https link. I did a factory reset and now I cannot connect through ADB securely. Using the link with the browser I get an error "Unable to connect", and with the terminal it fails to connect. 

[image]

[image]

If I open the port insecurely, I can connect through adb, meaning that both my pc and temi are in the same network. 

Why does it happen? How can I solve it?

Thanks

zjn0505, 2023-10-26:
The factory reset was made in temi settings or Android settings?

acivit, 2023-10-26:
In temi settings

acivit, 2023-10-26:
@zjn0505 In temi settings

polor6, 2024-01-02:
I encountered this problem in two ways before. The first one is to use adb devices to see how many devices you have. Usually when you encounter this kind of problem, it means that you did not give it a clear device name. Adb disconnect should solve the problem. The second is to shut down and restart both.The above is just a suggestion