robotemi/sdk #426: Is it ok to update android OS version on Temi ...?

YoshitakaOyama, 2023-11-10:
[question]
Can I update the Android version on Temi? Could it cause malfunction?
Also, if it is okay to update, how do I update the android version?

[background]
I'm currently trying out our web service on the Google Chrome app installed in Temi, but it's not working...

I also tried it on my Android tablet with latest OS version under the same conditions and it worked fine.
Therefore, I would like to update Temi's android version to the latest version and verify that our service works well.

YoshitakaOyama, 2023-11-11:
We are using microphone with [Web Speech API](https://developer.mozilla.org/en-US/docs/Web/API/SpeechRecognition). Our service is chatbot that supports voice input and output, and uses Web Speech API for voice recognition processing.

YoshitakaOyama, 2023-11-29:
@zjn0505 
Hi! 
I would appreciate any information regarding [this comment](https://github.com/robotemi/sdk/issues/426#issuecomment-1806935421).
Thank you!

YoshitakaOyama, 2023-11-29:
I also tried Web Speech API by [this google web page](https://www.google.com/intl/en/chrome/demos/speech.html) but I was not able to recognize audio too ...

zjn0505, 2023-11-29:
There are some problems in the WebView to use microphone and camera in the V3 tablet.

We are aware of this issue, but currently I cannot provide an estimation for the fix.

YoshitakaOyama, 2023-11-29:
I see. ok, thank you for your answering!

TarikOez, 2024-03-07:
Is there any new information regarding this issue? I've attempted to access the built-in microphone via a web view, but it seems not to be working.

zjn0505, 2025-04-17:
An update on this thread.

First please install a new system WebView.  e.g. https://www.apkmirror.com/apk/google-inc/android-system-webview/android-system-webview-134-0-6998-40-release/android-system-webview-134-0-6998-40-android-apk-download/

Then in you app with WebView.
```
    <uses-permission android:name="android.permission.RECORD_AUDIO" />
    <uses-permission android:name="android.permission.MODIFY_AUDIO_SETTINGS" />
```

These 2 permissions will be required and checked on runtime to use microphone.

and this for camera.
```
    <uses-permission android:name="android.permission.CAMERA" />
```