robotemi/sdk #427: How to trigger asr listener without using hey temi

mubashir-shk, 2023-11-17:
I am currently working on implementing a question-answer session with Temi. However, I want to eliminate the need for the user to say 'Hey Temi' before asking a question. I have two specific concerns:

1) How can I trigger the ASR listener without using the 'Hey Temi' command?
2) Will Temi always be in listening mode, or does it go to sleep after answering a question, similar to the behavior with 'Hey Temi'? I want to avoid the scenario where the user has to say 'Hey Temi' for each question, and instead, I want Temi to consistently listen and answer because users can ask questions at any time.

zjn0505, 2023-11-17:
`wakeup()` will make temi start to listen.

You will need to call wakeup() again to keep it listening, after TTS/ASR listeners giving you the event that conversation ends.

Perhaps add person detection listener to only start to listen when there is person in front.

Also your app can be kiosk app, so the converstaion layer can be overriden by your app.

You can look into this page https://github.com/robotemi/sdk/wiki/Speech for APIs you can use.

mubashir-shk, 2023-11-17:
Thank you so much for your reply. Just one more question: as the conversation ends after the first question, I will call wakeup() again. Suppose the user didn't instantly ask a second question and asked after 5 seconds, will Temi wait for the user until 5 seconds or will it go to sleep? How much time will Temi wait for user input on the ASR listener after calling wakeup()?