robotemi/sdk #432: AsrResult always returns SYSTEM sttLanguage?

LL469, 2023-12-28:
Hello,

I am using setAsrLanguages to have the robot be able to listen to multiple languages, and AsrResult picks up the text of the spoken language fine, but also always seems to return SYSTEM as the sttLanguage instead of the correct language being heard.
Does it work for anyone?

My Robox OS is 132.04, Launcher OS 17711 and app SDK version is 1.132.1

cyrus-rbc, 2024-01-02:
Hello @LL469 , we have confirmed that the first specified language is not correctly returned if SttLanguage.SYSTEM is not included in wakeup()/ setAsrLanguages(). We will fix this issue soon.

zjn0505, 2024-01-02:
Please submit a ticket at https://robotemi.zendesk.com/ and leave your robot's serial number there.

We will work on a fix and

closed 2024-01-02 by zjn0505