robotemi/sdk #437: Access to schedule of the robot (meeting and task)

jmLemaireEng, 2024-01-22:
Hi, I've been reviewing the SDK and it appears that there isn't a function allowing access to the robot's existing schedule. I haven't found any information regarding the meeting schedule or the scheduler with different sequences.

1.Is there a function for this that I might have missed?
          1.1. If not, is there a plan to include such a function in the future?
2.Is it possible for me to add events to the schedule?
       2.1. If yes, how can I do this?"
       2.2. If not, is there a plan to include such a function in the future?

zjn0505, 2024-01-22:
Meeting schedule and Scheduler are two different features in temi.

For the first one, you can schedule a meeting with `createLinkBasedMeeting` method.

But the other scheduler, there is no api to create one or get the list in temi SDK, and I cannot provide the estimation of adding it to SDK.

In my opinion this is a user oriented funciton to schedule a task in the future. On the other hand as developers we can use Android's AlarmManager to schedule a task in the app to start a sequence and run code with better flexibility.

jmLemaireEng, 2024-01-22:
Thank you for your prompt response. I am already aware of the createLinkBasedMeeting function, but I'm wondering if there's a way to access all the scheduled meetings. For example, if I want to create a connected calendar similar to Calendly but integrated with Temi, is this currently possible with the available API?

It's noteworthy that there isn't an API for the scheduler. This absence restricts the potential for enhancing Temi's intelligent behavior. Do you know if there are any plans to include this in the future?

The AlarmManager is intriguing, but it seems to require further development, as it necessitates essentially rebuilding the scheduler from Temi-Center, I believe.