robotemi/sdk #439: Temi always responds to user STT with same default message

BLG75, 2024-02-14:
Hello,

I am using the wakeup() method to have Temi listen for user speech, and the logic of my application is correctly using onAsrResult to acknowledge which key words the user input contains and trigger the appropriate goTo() method. However, Temi will not speak the appropriate message provided in my speak() method call. Instead he responds with the same default text no matter what the user says: "Check the temi app on your mobile device for your photos and videos" 
[image]
[image]

Ahad-Haque, 2024-03-17:
Hi, have you solved your issue?

zjn0505, 2024-03-18:
I suppose you haven't override temi‘s default NLP, so it is still working in the background.

onAsrResult() will give you the text user speaks, but it will not interrupt voice flow. So what user speaks will still be sent to temi NLP and converted into actions.

zjn0505, 2024-04-11:
https://github.com/robotemi/sdk/wiki/Speech#override-the-nlp-

This section describes how to override temi nlp

ShehryarZaheer, 2025-02-01:
Hi, does overriding the nlp also disables the Temi's functionality to respond to askQuestion queries?

zjn0505, 2025-02-03:
> Hi, does overriding the nlp also disables the Temi's functionality to respond to askQuestion queries?

Yes.

ShehryarZaheer, 2025-02-17:
So, how can we then ask questions to the users and listen to their input? Do we need to then use android speech recognition or Temi does provide another option to use the speech to text service?