robotemi/sdk #443: Looking for a robo temi developer for a POC project

ArnoldAtAchieva, 2024-03-02:
https://github.com/robotemi/sdk/blob/master/docs/sdk/com.robotemi.sdk/-robot/index.md 

I am looking to integrate a temi robot with a Tour Guide website. Temi robots can be programmed to follow the locations on a map, display live camera footage, show a Picture in a Picture video and play audio during the tour. https://www.youtube.com/watch?v=drupWULuJNE I want to create a simple POC website with two buttons: Tour 1 and Tour 2. When the Tour 1 button is pressed it should communicate to temi over the Internet and command it to follow the Tour 1 Sequence. I am looking to hire a robot developer to help me create the POC. 

https://github.com/robotemi

Please reach out to Arnold dot Villeneuve at achievatech dot com

arnoldvilleneuve, 2024-08-30:
No one interested in developing this project?

hncvj, 2025-01-19:
Interested. Are you still looking for a dev?

ArnoldAtAchieva, 2025-01-19:
It has been put on hold for budget reasons.

Arnold Villeneuve
MCT/MCE
CMMC PA/PI
Cloud Architect, Author, Instructor
613-833-0984