robotemi/sdk #446: To do all task one by one

ryanwyhk, 2024-03-18:
HI, 

I need to process below task one by one, but cannot success, even I added sleep  500, between them, Temi will perform all tasks. Should I use while loop to monitor the listeners to control the step flow? any advice or simply code reference please? Thank You

Process:
Step 1) robot.temiSay("Welcome")
Thread.sleep(500);
Step 2 )robot.temiSay("Follow me")
Thread.sleep(500);
Step 3)robot.GotoLocation(appointData.bookingRoom)

Ahad-Haque, 2024-03-18:
Hi, you can use sequence to achieve this. First you have to create one then call the sequence like playfirstsequence() to play your sequence.

ryanwyhk, 2024-03-18:
Thanks for your suggestion, but my process is dynamic not fixed, so I need use program to handle it

Ahad-Haque, 2024-03-18:
Then you can create dynamic sequence and play it.

ryanwyhk, 2024-03-18:
> Then you can create dynamic sequence and play it.
in Temi center set it? or use SDK has this sequence function?

zjn0505, 2024-03-19:
You can browse the wiki and sample app for each action you need.

It is important to note that you not only need to trigger the actions, but also to get their event callback, i.e. speech complete, goto location failure, etc.

Then you can chain the actions, and it will be something similiar to temi sequence feature.

ryanwyhk, 2024-03-19:
> You can browse the wiki and sample app for each action you need.
>
> It is important to note that you not only need to trigger the actions, but also to get their event callback, i.e. speech complete, [...]

Thank You, I try to search it

closed 2024-03-25 by ryanwyhk