robotemi/sdk #450: String from startDefaultNLU

Ahad-Haque, 2024-03-29:
How do I get string instead of Temi's voice from startDefaultNlu() function? or is their any other way of asking a question to Temi via text and get reply from temi via text also? Like chatgpt? but in this case Temi will be using its own NLP to process and reply user's text.

Thank you.

zjn0505, 2024-03-31:
There isn't such a way to use temi NLP like ChatGPT .

startDefaultNlu calls temi NLP and it will not always end up with a TTS response. 

Like "Save location as entrance" will trigger the save location intent, or a trained Q&A with responses in sequence or open an app.

Ahad-Haque, 2024-04-02:
Alright understand, then is there any way of getting reply (of a question) from Temi as Text and not voice?