robotemi/sdk #488: hwo to get Temi state

Sadre-abdelilah, 2025-02-24:
i'm trying to make an android app on my phone that checks what is temi doing (idle or moving to a location or charging etc), i couldnt find any function that can help me with that.

if it's idle i wanna perform an action so how can i check that.

zjn0505, 2025-02-25:
[OnGoToLocationStatusChangedListener](https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener)

Please check the listeners in this page. There is not active query but passive listener to get temi movement status.

Sadre-abdelilah, 2025-02-25:
i used it, but since we need to track if the robot is being used or not, i came across [onUserInteractionChangedListener ](https://github.com/robotemi/sdk/wiki/Detection-&-Interaction#onUserInteractionChangedListener) but ii didnt work for me, and i think its the best suit for our usecase, any suggestions on how to use ?

closed 2025-03-15 by Sadre-abdelilah