robotemi/sdk #494: Q&A through SDK

Sadre-abdelilah, 2025-03-12:
I wanna make Q&A like in the temi center but on my app, since i want the Questions and responses to be stored in my database, how can i do that ?

zjn0505, 2025-03-13:
To make Q&A in your app, the key is to take over temi's default NLP & Q&A, thus the user question goes to your app, rather than temi's own Q&A service.

You will need to 
1.  [override temi NLP](https://github.com/robotemi/sdk/wiki/Speech#override-the-nlp-) as a Kiosk app, this will make temi skip its default NLP.
2. Listen to `onAsrResult` callback, which provides the final speech to your app.
3. Then you can feed the question to your service/database for a best matched intent, then react accordingly.

Sadre-abdelilah, 2025-03-14:
Thank you, the issue im facing now is in activating kiosk mode, can i know what im doing wrong, following the [kiosk mode guide](https://github.com/robotemi/sdk/wiki/Kiosk-Mode) i still cant get it activated, no popup is shown, can i please know what we are doing wrong ? 
in androidManifest :
<meta-data
            android:name="com.robotemi.sdk.metadata.SKILL"
            android:value="@string/app_name" />

        <meta-data
            android:name="com.robotemi.sdk.metadata.KIOSK"
            android:value="TRUE" />

        <meta-data
            android:name="@string/metadata_override_conversation_layer"
            android:value="true" />

and heres the mainActivity  :
public class MainActivity extends AppCompatActivity {

    private Robot robot;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        robot = Robot.getInstance();

        // Request Temi to set this app as Kiosk app
        robot.requestToBeKioskApp();

        // Check if Kiosk Mode is activated
        if (robot.isKioskModeOn()) {
            Toast.makeText(this, "Kiosk mode activated successfully!", Toast.LENGTH_LONG).show();
            Log.d("TemiKiosk", "Kiosk mode is active.");
        } else {
            Toast.makeText(this, "Kiosk mode is not active yet. Please approve it on Temi.", Toast.LENGTH_LONG).show();
            Log.w("TemiKiosk", "Kiosk mode is NOT active.");
        }

    }

zjn0505, 2025-03-17:
You don’t have to override conversation layer. Only override that when you don’t need temi default conversation layer which is the pop up in your case any more.

Sadre-abdelilah, 2025-03-17:
yes, thats my need, but i did remove it and still not activated, as you can see i use a Toast to check its activation, is everything else enough ? and do we need to install the app for it to activate kiosk mode

Sadre-abdelilah, 2025-03-17:
it succeeded to use the kiosk mode, but i need as you know to answer with my iwn answers, so i used [AsrListener](https://github.com/robotemi/sdk/wiki/Speech#asrListener) but the answers are still from default temi :
@Override
    public void onAsrResult(String asrResult, SttLanguage sttLanguage) {
        if (asrResult == null || asrResult.trim().isEmpty()) return;

        String recognizedText = asrResult.trim().toLowerCase();
        Log.d("TemiSpeech", "User said: " + recognizedText + ", language: " + sttLanguage);

        // 4) Look up the recognized text in your Q&A map
        String answer = qaMap.get(recognizedText);

        if (answer != null) {
            // Found an answer in our dictionary
            robot.speak(TtsRequest.create(answer, false));
        } else {
            // Not found - fallback response
            robot.speak(TtsRequest.create("I'm sorry, I don't have an answer for that.", false));
        }
    }

also in the manifest heres what im using : 
<meta-data
            android:name="com.robotemi.sdk.metadata.SKILL"
            android:value="@string/app_name" />

        <meta-data
            android:name="com.robotemi.sdk.metadata.KIOSK"
            android:value="TRUE" />
        <meta-data
            android:name="com.robotemi.sdk.metadata.NLU"
            android:value="true" />

zjn0505, 2025-03-18:
In the meta-data it should be

`com.robotemi.sdk.metadata.OVERRIDE_NLU`

or use `@string/metadata_override_nlu`