robotemi/sdk #495: Kiosk mode not activated

Sadre-abdelilah, 2025-03-17:
Hi, we are trying to open Kiosk mode, but for some reason it doesn't activate, can you please check if we are doing it right : 
heres the androidManifest : 
`<meta-data
            android:name="com.robotemi.sdk.metadata.SKILL"
            android:value="@string/app_name" />

        <meta-data
            android:name="com.robotemi.sdk.metadata.KIOSK"
            android:value="TRUE" />
        <meta-data
            android:name="com.robotemi.sdk.metadata.NLU"
            android:value="true" />`

and heres the code : 
`public class MainActivity extends AppCompatActivity implements AsrListener {

    private Robot robot;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        robot = Robot.getInstance();

        // 1) Request kiosk mode so your app overrides default NLP.
        robot.requestToBeKioskApp();

        // 2) Register AsrListener for direct speech text + language.
        robot.addAsrListener(this);

        // Check kiosk mode status
        if (robot.isKioskModeOn()) {
            Toast.makeText(this, "Kiosk mode is active.", Toast.LENGTH_LONG).show();
            Log.d("TemiKiosk", "Kiosk mode is active.");
        } else {
            Toast.makeText(this, "Kiosk mode is NOT active. Approve on Temi screen.", Toast.LENGTH_LONG).show();
            Log.w("TemiKiosk", "Kiosk mode is NOT active.");
        }

        // Greeting
        robot.speak(TtsRequest.create("Hello, I'm listening.", false));
    }`

As i saw in the [kiosk mode](https://github.com/robotemi/sdk/wiki/Kiosk-Mode) i should see a new option in the settings called Kiosk Mode, tried also to install the app and choose it as home screen but still nth.

zjn0505, 2025-03-17:
I guess when you call the robot method in onCreate callback, it might be likely that the robot is not ready, meaning the service binder to talk to temi SDK service has not been established yet. You can move the robot calling methods to onRobotReady callback and if there is any difference

Sadre-abdelilah, 2025-03-17:
Thank you, that works.

closed 2025-03-17 by Sadre-abdelilah