robotemi/sdk #499: Moving head and the base at the same time

jocacace, 2025-03-31:
Hello,

I am trying to control at the same time the tilt angle of the head and the base velocity using the SDK calls. I am trying to implement something like the follow mode.

However, when I try to send both commands almost in the same time one of the two is ignored (I should stop the motions and ask for a tilt angle motion later). Is this the normal way in which it works? In this case, are there any other solution for my issue?

Regards,

zjn0505, 2025-04-01:
There can be only one onging movement command, one will overtake another if sent together.

The tilt angle of the tablet is controlled by the system when moving, since therer are depth cameras on the tablet which will be used for obstacle detection.

jocacace, 2025-04-01:
Thanks, however would be nice to allow the user to take the control of the camera during the motion as well.

closed 2025-04-01 by jocacace